radhoo/uradmonitor_kit1 #3: can not remove "upgraded" stations from uradmonitor map

twanek, 2016-10-16:
**simptome:**
when flashing the new firmware, if one does not set the "do not clear eeprom" fusebit parameter, the server allocates a new device id.
so, every time when the firmware is flashed, a "new' device will appear on the map, and the "old" ones still remain. this can clutter the map and could suggest improper device numbers / city. posteriorly impossible for the user to remove them...

**solution:**
it would be nice to have a delete button on the dashboard (or some other way) to delete fictional stations from the map. (or delete automatically the units which are offline more than x days)

jetty840, 2017-06-14:
I've also had it happen once when the fuse is set correctly, but the upload comms corrupted.  Delete feature would be good.

radhoo, 2017-06-26:
The solution was implemented on the server side . ID of "Old" units are being reallocated to new requesters. An ID is labeled "Old" if more than 30 days have passed since the last data was received.

closed 2017-06-26 by radhoo