radhoo/uradmonitor_kit1 #13: Resistor value changes to reduce overall current consumption

Wolferl1, 2017-07-11:
I'd like to suggest some resistor value changes which allow for reduced overall current consumption of the KIT1 without impairing its performance. I tested all if these changes successfully on my KIT1.

1. The base resistor R9 to T4 for the LED backlight is IMHO much too low with 100 ohms.
I changed it to 2k7, and T4 still saturates to 0,05V when the backlight LEDs are on. I assume a 4k7 resistor will be just fine here.

2. The same goes with the base resistor R10 for T5 (beeper). I also used a 2k7 here (and could also easily be 4k7). There is no noticable decrease of beeper sound volume.

3. With the standard values, the pulse width for SBM20 tube anode voltage generation was about 64% (whole unit needed 42mA at 5V). The voltage divider R3 and R4 could be made better with 47Mohms and 220kOhms, greatly reducing the load here.

4. R1's value is way too low. Originally, I scoped a base current of 15mA into T1 (when T1 is turned on). For an estimated "on-state" collector current of 100mA and a MPSA42 has a hFE of at least 40, so a base current of about 2mA should be ok. That gives a base resistor value of about 2k.
I tried successfully a value of 1k5 instead of 100 ohms.

All these changes do not require a software modification.

Doing this, the HV converter pulse width reduced to about 43%, and the unit only draws 25mA now.

radhoo, 2017-08-05:
Thank you. Here are the modifications fixing these issues in 1.2.106 and my comments:
1. R9 changed to 4.7K
2. R10 changed to 4.7K
3. I used 22M/100K for R3 and R4. Centronics Geiger tube theory recommends a certain ratio between these two, that is also important for getting the right pulse amplitude for the following blocks. The change 22M/100K didn't change the duty load. So I prefer to go with 10M/47K as these are more common components.
4. R1 was changed to a multitude of values: 180, 330, 470, 1K, 2.2K and the results were bad. This was kept to the original 100Ohm value.

With these changes, the pulse width was reduced to 43% and on some units to 40%. Please keep in mind that this is not the duty cycle that the switching transistor sees. The switching transistor, MPSA42, sees a reduced duty cycle due to the blocker PNP transistor used to collapse the magnetic field faster, and so obtaining higher amplitude spikes across the inductor, needed to make the jump from 3.3V to 380V. Still this improvement is important, as some inductors were at their upper limit for providing the needed voltage and now the margin is bigger.

The power consumption also improved negligibly. 

[image]

closed 2017-08-05 by radhoo

Wolferl1, 2017-08-05:
Hi Radu,
Thank you for reviewing this. I agree with you at points 1 and 2.
With point 3, I beg to differ *g*
The resistor divider network 10M / 47k is only for sensing the tube voltage. This 10M is IMHO an unnecessary high load. I now use 47M and 220k with very good results. Don't confuse this with the 2nd 10M resistor which is the anoder working resistor for the tube. This resistor has already the correct 10M value.
However I do agree with you that 22M or even 47M are bigger (especially the 500V capable ones) and are not that common.

As for point 4, it could be that you needed lower values for R1 because the load on the HV line is high than on my KIT1.

Kind regards,
Wolferl

radhoo, 2017-08-05:
Sorry I wrote this in a hurry and did a mistake. R3/R4 is for the voltage feedback. Allow me to edit the answer to avoid any confusions.

radhoo, 2017-08-05:
Done. Your contribution helped this design a lot. Getting the duty to 40-43% solves the issue some other builders where having identifying the exact coils I've used.
For point 4 I also agree with your input, and was prepared to change it as well, then I did some tests on 2-3 devices and I was unable to get it right. More testing will be needed.  Values close to 100Ohm worked fine, but getting aways quickly created problems.