radhoo/uradmonitor_kit1 #17: Missing Documentation and BOM

mnemocron, 2020-11-21:
The project in itself is a very neat thing to build. But can you please add more documentation.
Just from the files alone I was not able to order the correct parts.

The schematics do not give away the exact part-nr. for the components.
This is tricky not only for capacitors but mostly for the electromechanical components like the fuse-clips or the switch.

Furthermore, I managed to order the wrong LCD module - the one from Sparkfun - as opposed to the cheaper Chinese one with a _different pinout_.
Then I ordered a 1117 Regulator as a THT part because many images online contain a THT regulator.
But the most recent revision of the PCB (`v1.2.106`) apparently features an SMT regulator.

.. I hope you see my point.
**Please add at least a proper BOM** with reference to the hardware revision (eg. `v1.2.106`) 

Here is a start. This is what I ordered from digi-key

[https://octopart.com/bom-tool/xPA7yJNx](https://octopart.com/bom-tool/xPA7yJNx)
> (note: that the BOM is incomplete and does contain extra parts and wrong parts)

---

Another part you should document is, how you compile and upload the code to the board.
The URL in the `README` is outdated and does not work anymore.
I used the following commands in WSL to compile the code:

```bash
sudo apt install make avr-gcc avr-libc gcc-avr
cd ./code
make
```

and the following commands in PowerShell to upload the code to the board using ASP

```powershell
& 'C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avrdude' -CC:'\Program Files (x86)\Arduino\hardware\tools\avr/etc/avrdude.conf' -v -patmega328p -cusbasp -Pusb -Uflash:w:C:\users\simon\Downloads\uRADMonitor.hex:i
```
or short
```powershell
avrdude.exe -Cavrdude.conf -v -patmega328p -cusbasp -Pusb -Uflash:w:uRADMonitor.hex:i
```

--- 

And maybe you could document how the Software works.
What the LCD displays and what the button does.
This is actually covered in the Youtube Video linked somewhere in the repo.

radhoo, 2020-11-21:
Hello Simon,

Thanks for the nice feedback.

There are a few articles on the uRADMonitor blog that bring in the missing pieces, see: https://www.uradmonitor.com/blog/?s=kit1

There was also a article on how to compile the KIT1 code  but it got deleted with no recovery due to an unexpected situation.

If interested feel free to give back to the community yourself, and write a quick guide or even a blog post on the uRADMonitor blog.

Let me know if interested, that would help others like you with what you learned so far.

mnemocron, 2020-11-21:
Thanks for the hint with the forum. When I was researching my BOM, I quickly left the forum website as it did not seem to contain the information that I need (still doesn't). 

I just looked at the articles you linked.
The issue with the articles is: they do not contain a proper BOM.
They either link back to this repo on Github or they contain some sort of "tutorial" for an outdated hardware.

But there is a post that describes how to program it at least (using USB ASP): 
[https://www.uradmonitor.com/open-source-uradmonitor-kit1/](https://www.uradmonitor.com/open-source-uradmonitor-kit1/)

And a post describing what the firmware does:
[https://www.uradmonitor.com/model-kit1-production-ready/](https://www.uradmonitor.com/model-kit1-production-ready/)

I should not have to research a BOM from some Forum posts that are not linked inside the original repo.

So yes, I would like to share my findings as soon as I have a Kit running. (In a way I already did here)
Unfortunately I now need to order again... 
It just feels odd that you encourage me to reverse engineer something that you already know.

IkaSGC, 2020-11-21:
If someone could bring quality instructions on how to put it together and flash using windows, it would be amazing. I have theoretically materials for several uRad monitors at home. I've passed one so far, but I can't get anything out of it. Even the display shows nothing. Obviously an error occurred somewhere, but I can't figure it out with the little information I found.

radhoo, 2020-11-21:
Also see: https://www.uradmonitor.com/topic/opensource-kit1-firmware/ and the rest of the forum.

radhoo, 2020-11-21:
For compiling on Windows there already is a tutorial here: https://www.uradmonitor.com/a-tutorial-how-to-compile-kit1-firmware-on-a-windows-machine/

mnemocron, 2020-12-26:
a wonderful good morning everyone,

I have finished up the BOM on Octopart.
I used the [bom.html](https://github.com/radhoo/uradmonitor_kit1/blob/master/pcb/1.2.106/bom.html) from the original repository and added actual part numbers (for digikey).

It would be helpful if you could copy / paste the below markdown BOM to your repos Wiki or Readme.md

BOM on [Octopart](https://octopart.com/bom-tool/xPA7yJNx)

BOM as Markdown:

| Qty | Line Item          | Manufacturer        | MPN                | Schematic Reference | Supplier                                            | Description                                                           | SKU                     | Unit Price (@ Batch Size: 2) | URL                                                                                                    | Unit Price (@ Batch Size: 2) [Lowest Price   (Overall)] | Line Total [Lowest Price (Overall)] |
|-----|--------------------|---------------------|--------------------|---------------------|-----------------------------------------------------|-----------------------------------------------------------------------|-------------------------|------------------------------|--------------------------------------------------------------------------------------------------------|---------------------------------------------------------|-------------------------------------|
|     |                    |                     |                    |                     |                                                     |                                                                       | Digi-Key                | Digi-Key                     | Digi-Key                                                                                               |                                                         |                                     |
| 1   | 1N4148TR           | Vishay              | 1N4148TR           | D7                  |                                                     | Diode, Small Signal; 300 mA; 100 V                                    | 1N4148VSCT-ND           | 0.08896                      | https://octopart.com/click/track?ai=8237&sig=09ebca2&sid=459&ppid=73&vpid=188125481&ct=offers          | 0.004003                                                | 0.004003                            |
| 3   | PN2222ATA          | ON Semiconductor    | PN2222ATA          | T2, T4, T5          |                                                     | NPN 625 mW 40 V 1 A Through Hole                                      | PN2222ATACT-ND          | 0.204608                     | https://octopart.com/click/track?ai=8237&sig=0d3e9e8&sid=459&ppid=84409647&vpid=452472732&ct=offers    | 0.119307                                                | 0.357922                            |
| 2   | PN2907ATAR         | ON Semiconductor    | PN2907ATAR         | T3, T6              |                                                     | 60 V 800 mA Through Hole PNP                                          | PN2907ATARCT-ND         | 0.231296                     | https://octopart.com/click/track?ai=8237&sig=083ef36&sid=459&ppid=84409654&vpid=452472745&ct=offers    | 0.032292                                                | 0.064585                            |
| 1   | ATMEGA328P-PU      | Microchip           | ATMEGA328P-PU      | ATMEGA328P          | (with bootloader flashed) ebay.com / aliexpress.com | 8 Bit MCU (Arduino Bootloader optional)                               | ATMEGA328P-PU-ND        | 1.850368                     | https://octopart.com/click/track?ai=8237&sig=04b47fc&sid=459&ppid=77760224&vpid=408716895&ct=offers    | 1.75607                                                 | 1.75607                             |
| 3   | BYV26E-TR          | Vishay              | BYV26E-TR          | D1, D2, D3          |                                                     | Diode, Ultrafast; 2.5 V                                               | BYV26ECT-ND             | 0.498176                     | https://octopart.com/click/track?ai=8237&sig=02c1c43&sid=459&ppid=39446776&vpid=189200847&ct=offers    | 0.148563                                                | 0.44569                             |
| 1   | MPSA42             | ON Semiconductor    | MPSA42             | T1                  |                                                     | NPN 625 mW 300V 500mA Through Hole                                    | MPSA42FS-ND             | 0.391424                     | https://octopart.com/click/track?c=1&ai=8237&sig=00c6b34&sid=459&ppid=5050&vpid=452471303&ct=offers    | 0.048661                                                | 0.048661                            |
| 4   | CFR-25JB-52-10K    | Yageo               | CFR-25JB-52-10K    | R2, R8, R14, R15    |                                                     | 10kOhm 1/4W 5% Axial                                                  | 10KQBK-ND               | 0.08896                      | https://octopart.com/click/track?ai=8237&sig=0e1eb73&sid=459&ppid=22569022&vpid=39633415&ct=offers     | 0.08896                                                 | 0.35584                             |
| 2   | CFR-25JB-52-10M    | Yageo               | CFR-25JB-52-10M    | R3, R5              |                                                     | 10MOhm 1/4W 5% Axial                                                  | 10MQBK-ND               | 0.08896                      | https://octopa [... cut ...]                      |
| 1   | EG1218             | E-Switch            | EG1218             | S1                  |                                                     | Toggle Switch (not sideways)                                          | EG1903-ND               | 0.57824                      | https://octopart.com/click/track?ai=8237&sig=0e9adab&sid=459&ppid=1056201&vpid=1179261&ct=offers       | 0.21797                                                 | 0.21797                             |

mnemocron, 2020-12-26:
@IkaSGC just an idea. Which LCD Modules do you have? 
I actually bought the more expensive Sparkfun modules. But they have the wrong pinout, different from the Chinese ones you find on aliexpress. When I rewired the LDC module with the help of the schematics and some breadboard wires it worked.

IkaSGC, 2020-12-26:
@mnemocron  I bought these modules.

https://www.aliexpress.com/item/32526061324.html?spm=a2g0s.9042311.0.0.27424c4dxcoCrr